tuanchris/dune-weaver #1: Docker build error

Eng-7, 2024-12-17:
I've tried to use the docker compose file to create a container but unfortanly I get following error:

docker compose up --build -d
[+] Running 0/1.2s (7/8)                                                                                        docker:default
[+] Building 90.5s (7/8)                                                                                        docker:default 
[+] Building 115.7s (8/8) FINISHED                                                                              docker:default 
 => [flask-app internal] load build definition from Dockerfile                                                            0.4s 
 => => transferring dockerfile: 524B                                                                                      0.0s 
 => [flask-app internal] load metadata for docker.io/library/python:3.9-slim                                              1.4s 
 => [flask-app internal] load .dockerignore                                                                               0.1s 
 => => transferring context: 2B                                                                                           0.0s 
 => [flask-app 1/4] FROM docker.io/library/python:3.9-slim@sha256:4ee0613170ac55ebc693a03b6655a5c6f387126f6bc3390e739c2e  0.0s 
 => [flask-app internal] load build context                                                                               0.1s 
 => => transferring context: 5.07kB                                                                                       0.0s 
 => CACHED [flask-app 2/4] WORKDIR /app                                                                                   0.0s 
 => CACHED [flask-app 3/4] COPY . /app                                                                                    0.0s 
 => ERROR [flask-app 4/4] RUN pip install --no-cache-dir -r requirements.txt                                            112.6s                                                                                                                         
 > [flask-app 4/4] RUN pip install --no-cache-dir -r requirements.txt:                                                         
13.31 Collecting flask                                                                                                         
13.61   Downloading flask-3.1.0-py3-none-any.whl (102 kB)                                                                      
13.68      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 103.0/103.0 kB 1.6 MB/s eta 0:00:00                                         
13.81 Collecting pyserial                                                                                                      
13.83   Downloading pyserial-3.5-py2.py3-none-any.whl (90 kB)
13.85      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 90.6/90.6 kB 20.4 MB/s eta 0:00:00
17.48 Collecting numpy
17.51   Downloading numpy-2.0.2.tar.gz (18.9 MB)
19.88      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 18.9/18.9 MB 8.0 MB/s eta 0:00:00
40.40   Installing build dependencies: started
88.63   Installing build dependencies: finished with status 'done'
88.64   Getting requirements to build wheel: started
89.71   Getting requirements to build wheel: finished with status 'done'
89.74   Installing backend dependencies: started
94.83   Installing backend dependencies: finished with status 'done'
94.83   Preparing metadata (pyproject.toml): started
104.6   Preparing metadata (pyproject.toml): finished with status 'error'
104.6   error: subprocess-exited-with-error
104.6   
104.6   × Preparing metadata (pyproject.toml) did not run successfully.
104.6   │ exit code: 1
104.6   ╰─> [20 lines of output]
104.6       + /usr/local/bin/python3.9 /tmp/pip-install-0l1ag2j8/numpy_23b8d891a1cd4799841286688937d8fd/vendored-meson/meson/meson.py setup /tmp/pip-install-0l1ag2j8/numpy_23b8d891a1cd4799841286688937d8fd /tmp/pip-install-0l1ag2j8/numpy_23b8d891a1cd4799841286688937d8fd/.mesonpy-ae0vvoaz -Dbuildtype=release -Db_ndebug=if-release -Db_vscrt=md --native-file=/tmp/pip-install-0l1ag2j8/numpy_23b8d891a1cd4799841286688937d8fd/.mesonpy-ae0vvoaz/meson-python-native-file.ini
104.6       The Meson build system
104.6       Version: 1.4.99
104.6       Source dir: /tmp/pip-install-0l1ag2j8/numpy_23b8d891a1cd4799841286688937d8fd
104.6       Build dir: /tmp/pip-install-0l1ag2j8/numpy_23b8d891a1cd4799841286688937d8fd/.mesonpy-ae0vvoaz
104.6       Build type: native build
104.6       Project name: NumPy
104.6       Project version: 2.0.2
104.6       
104.6       ../meson.build:1:0: ERROR: Unknown compiler(s): [['cc'], ['gcc'], ['clang'], ['nvc'], ['pgcc'], ['icc'], ['icx']]
104.6       The following exception(s) were encountered:
104.6       Running `cc --version` gave "[Errno 2] No such file or directory: 'cc'"
104.6       Running `gcc --version` gave "[Errno 2] No such file or directory: 'gcc'"
104.6       Running `clang --version` gave "[Errno 2] No such file or directory: 'clang'"
104.6       Running `nvc --version` gave "[Errno 2] No such file or directory: 'nvc'"
104.6       Running `pgcc --version` gave "[Errno 2] No such file or directory: 'pgcc'"
104.6       Running `icc --version` gave "[Errno 2] No such file or directory: 'icc'"
104.6       Running `icx --version` gave "[Errno 2] No such file or directory: 'icx'"
104.6       
104.6       A full log can be found at /tmp/pip-install-0l1ag2j8/numpy_23b8d891a1cd4799841286688937d8fd/.mesonpy-ae0vvoaz/meson-logs/meson-log.txt
104.6       [end of output]
104.6   
104.6   note: This error originates from a subprocess, and is likely not a problem with pip.
104.6 error: metadata-generation-failed
104.6 
104.6 × Encountered error while generating package metadata.
104.6 ╰─> See above for output.
104.6 
104.6 note: This is an issue with the package mentioned above, not pip.
104.6 hint: See above for details.
105.5 
105.5 [notice] A new release of pip is available: 23.0.1 -> 24.3.1
[+] Running 0/1To update, run: pip install --upgrade pip
 ⠙ Service flask-app  Building                                                                                          116.1s 
failed to solve: process "/bin/sh -c pip install --no-cache-dir -r requirements.txt" did not complete successfully: exit code: 1

Used Hardware:

Raspberry Pi 3B+
with the image 2024-11-19-raspios-bookworm-armhf-lite.img

Thanks for help! :)

tuanchris, 2024-12-17:
Strange, that should work for everyone. Let me test when I get back. Meanwhile, you can run python app.py and see if everything works.

Eng-7, 2024-12-17:
i ran directly the app.py and i got it working. I used following to get it running:

`pip list
Package            Version
------------------ ---------
asgiref            3.6.0
Babel              2.10.3
blinker            1.5
certifi            2022.9.24
chardet            5.1.0
charset-normalizer 3.0.1
click              8.1.3
colorama           0.4.6
colorzero          2.0
cryptography       38.0.4
distro             1.8.0
Flask              2.2.2
gpiozero           2.0.1
idna               3.3
itsdangerous       2.1.2
Jinja2             3.1.2
lgpio              0.2.2.0
MarkupSafe         2.1.2
numpy              1.24.2
pigpio             1.78
pip                23.0.1
pycryptodomex      3.11.0
pyinotify          0.9.6
pyOpenSSL          23.0.0
pyserial           3.5
python-apt         2.6.0
python-dotenv      0.21.0
pytz               2022.7.1
requests           2.28.1
rpi-lgpio          0.6
setuptools         66.1.1
simplejson         3.18.3
six                1.16.0
smbus2             0.4.2
spidev             3.5
ssh-import-id      5.10
toml               0.10.2
urllib3            1.26.12
Werkzeug           2.2.2
wheel              0.38.4`

The versions are different to the ones from the container.

closed 2024-12-22 by tuanchris