tuanchris/dune-weaver #15: Https support (load as iframe in Home Assistant)

Thokoop, 2025-01-14:
I would like to have the webapp running via https, so the web app can be loaded in an iframe in Home Assistant Dashboards

ProtoxiDe22, 2025-01-15:
This seems like a good idea, however what i think could even be better is directly integrate the api so that you can see entities in HA.

This shouldn't be too hard using MQTT autodiscovery. i'll try to get more info on this when i got a bit of time and report back

Thokoop, 2025-01-15:
That would be even better indeed. Than we'd are able to use it also in automations 😄

ProtoxiDe22, 2025-01-15:
yes that's the point! I plan to sync my table to the living room lights, it will turn on when the lights are on, i planned to do this with the api, but integrating the duneweaver into HA directly would be next level

ProtoxiDe22, 2025-01-15:
Yeah that would be needed. but it does not seem too hard to do honestly.
I still haven't had time to look at the backend code but will try to do it soon. Never used Flask, but I work as a django developer during the day so shouldn't be too hard :)

closed 2025-02-28 by tuanchris