tuanchris/dune-weaver #42: Integrate backend into the ESP32

BenJuan26, 2025-02-18:
Cool project!

I'm going to build a Mini, as I already have nearly all the required hardware. I got thinking, though: how much work would it be to have the ESP32 handle both the motor logic as well as the webapp logic? If that were the case, the table would be completely self-contained, only needing power and the correct Wifi credentials.

It's likely that I'll look into it myself, but of course the maintainers will know better than me whether the backend is too complex to fit into the memory of the ESP32.

ProtoxiDe22, 2025-02-18:
This would require major work, or rather a complete rewrite of everything.

The entire flask project that runs on the Pi would have to be rewritten in C in order to be able to run on the esp32. and that poses other challenges too:
First of all: for all the other versions that are not the Mini, we are moving towards using esp32 or arduino with fluidnc/grbl firmware. dropping that would require us to rewrite the driver handling code, which we tried initially and it's not easy at all.
There is also the memory problem: the pattern files are kinda big, some of them are big enough that the whole esp32 flash memory is not enough to store them. So this means that it would be mandatory to have an SD module where to store the patterns.
I don't think this is realistically feasible. We are looking at ways to make the installation/configuration process easier, but  as far as hardware goes i think the Pi will always be needed.

ProtoxiDe22, 2025-02-18:
That's being worked on! join the discord server as that's where it's being discussed if you want to be involved

closed 2025-02-18 by BenJuan26