tuanchris/dune-weaver #54: help to proceed

zorro1011, 2025-04-26:
hi
first of all thank you for sharing, with all of us, your projects and compliments.

maybe I made a mistake in starting the dune_weaver_mini project, but I had all the electronics and I only needed to print the pieces with my 3D printer.
now that I have switched to the software I am in huge difficulty because I have not understood what to do.
I loaded the FW on the esp32 from fluidNC and I tried the BT version first.
I hoped to be able to do everything from the phone with BT, but despite having connected successfully I was unable to do anything.
then I tried the WIFI one by configuring it as AP, but at most I managed to access a test panel from a PC.

now the question:
is there a simple way to load the .thr files on the esp and have them draw?
if it is not possible, can I ask what is the way to do it. (I apologize for my unpreparedness :-( )

thanks in advance
best regards
zorro

zorro1011, 2025-04-27:
After an hour with Copilt AI, I managed to get it to work by installing Python on Win11.
The web page is fantastic and even shows previews of the drawings.
I am speechless.

THANK YOU so much for the great work done and I apologize again for not having figured out by myself how to proceed to make it work.

it would be great if everything could be done with an app on the phone

now I offer you a coffee, it seems like the least I could do

thanks
best regards

closed 2025-04-30 by tuanchris

tcsabina, 2025-05-02:
hi @zorro1011,

could you share your steps to solve this?
I am in a similar struggle: reading the wiki, but it is not clear why I need to do what is written in there...

I have flashed the esp32 from fluidNC, and the next step on the Wiki is "Deploying backend code". But I have no clue what is that step about? What backend code? What webpage? What should I do on a RPI or on a PC?
On the PC that is connected to the esp32? Or some other PC?